OpenWaterProject/riffle_328 #14: licence

vjpixel, 2015-07-27:
on the top level I found the CERN licence, but the licence says "2.3. This Licence does not extend to software, firmware, or code loaded into programmable devices..."

I'm working in a project that use some code from Riffle, so I want to choose the right licence.
https://github.com/InfoAmazonia/water-sensor
https://github.com/InfoAmazonia/rede-hardware

jywarren, 2015-07-28:
Hi, vjpixel! @ebarry pointed me here. Where is the Riffle code you're using?

vjpixel, 2015-07-28:
sorry, I got the wrong info from the team here. we're not using code from Riffle, we're using Coqui [0] as part of the hardware.

[0] http://publiclab.org/notes/donblair/09-30-2014/coqui-bbv1-0

@MarceloPesse

jywarren, 2015-07-28:
Hi, @vjpixel and @marcelopesse - well, at the bottom of every page there are licenses at the foot of the page, including the CERN OHL. However, the clause you mention in the CERN OHL just means that any software that is used to *run* the hardware must have its own license. The Coqui doesn't use software or firmware or code on-board, so it's not relevant here. But yes - you'll have to use CERN OHL if you've based your hardware work on the Coqui.

closed 2015-12-09 by dwblair