OpenWaterProject/riffle_328 #26: Fix labeling on protoboard

dwblair, 2016-09-09:


pdhixenbaugh, 2018-05-03:
> Doh! Still an issue, unfortunately!

Is there currently work being done on this model of Riffle?

shapironick, 2018-05-03:
I’m not sure about dev work, but I was attempting to use it to log some air sensor data but the board labeling scuttled that work once we got to the field. 

Sent on the move

jywarren, 2018-05-04:
Hi, I don't think there are any of these left here, but here's one I found: 

[image]

shapironick, 2018-05-04:
Here is what Kina wrote:

Basically, the rows need to be inverted.

If they looked like this:

                 ------Pins are here------
A0        SCL    SDA    D2         D3         3V3      GND
VBAT   A1       A2       SCK      MOSI     MISO   D10

Actual connections are REVERSED. like so:

                 ------Pins are here------
VBAT   A1       A2       SCK      MOSI     MISO   D10
A0        SCL    SDA    D2         D3         3V3      GND