OpenWaterProject/riffle_328 #30: Evaluate USB-Serial alternatives -- find easier-to-source chip than CH340

dwblair, 2016-09-09:


jywarren, 2016-09-09:
Hey Don!! is this for writing to memory or for programming?

Have you seen webjack? publiclab/webjack

dwblair, 2016-09-10:
Jeff!  This is for programming ... but yes, webjack looks cool!

Because of weird power fluctuation issues with SD cards, I've actually been thinking about storing data on eeprom on the board and reading it out via usb .... and that mode would make webjack especially appealing.  Does it work pretty solidly now?

jywarren, 2016-09-12:
There are certain requirements -- atmega processor (though attinys can
write, just not read yet), but yeah, check out Richard's demo!
https://publiclab.org/tag/webjack

dwblair, 2016-09-14:
Just spent the last two days playing with WebJack.  Wow, it's fun to get sensor readings in the browser :)

I'll post a research note soon, but this is a Pro Mini + a thermistor + rickshaw.js :
https://www.youtube.com/watch?v=GtJW1Dlt3cg

Wheee!  Brilliant work on that project.

pdhixenbaugh, 2016-09-15:
Pretty fun! I can almost imagine a webapp that works on any phone... Now I
hope the analog audio interface survives for a pretty long time.

dwblair, 2016-09-15:
Haha, I know -- iPhones audio input is going digital ... so who knows what'll happen to our beloved audio jack :)  But, I think the nice thing with the WebJack approach is:  it just needs to have access to audio input *somehow* -- whether that's coming in through wires, or wireless.  Even if we have to make a little speaker that chirps into the phone / laptop mic,  we can be sure that audio input to the computer of *some* sort is going to remain a feature, as long as we want to be able to talk to these devices (until Apple releases iCranium wireless brainternet chips).

jywarren, 2016-09-15:
Then we'll release BrainJack, an open source JS-to-thought API.

Rad video, thanks!!!